D-314/IP2368-Arduino-Library #4: IP2366

lAbenl, 2024-06-02:
Is the Library works on IP2366?

D-314, 2024-06-11:
This library was tested only on IP2368. I think they should be compatible, but it's better to check it out. Could you provide a register card for IP2366, please?

Eddiiie, 2024-08-20:
I have the datasheet translated.  It has some register information in there...   A bit misfoRmaTTed.
[ip2366.pdf (1).pdf](https://github.com/user-attachments/files/16667148/ip2366.pdf.1.pdf)

Using a Pi, I'm able to scan the I2C bus and get the address back, but not much more than that..

D-314, 2024-08-22:
@Eddiiie, thanks for the register map for IP2366.
It looks like IP2366 has the following changes relative to IP2368:
| 🚩 |  action  | addr |                                             register description                                              |                                          comment                                          |
|----|----------|------|---------------------------------------------------------------------------------------------------------------|-------------------------------------------------------------------------------------------|
| ⚠️ | modified | 0x00 | SYS_CTL0 (charge enable register)                                                                             | do not have En_Vbus_Sinkctrl flag                                                         |
| ❌  | deleted  | 0x01 | SYS_CTL1 (series number setting, battery type, current setting mode)                                          |                                                                                           |
| ⚠️ | modified | 0x02 | SYS_CTL2 (Vset full charge voltage setting)                                                                   | do not have Enable Flag; lowest voltage now is 2.5V                                       |
| ⚠️ | modified | 0x03 | SYS_CTL3 (Iset charging power or current setting)                                                             | do not have Enable Flag                                                                   |
| ❌  | deleted  | 0x04 | SYS_CTL4 (battery capacity setting)                                                                           |                                                                                           |
| ❌  | deleted  | 0x06 | SYS_CTL6 (current battery power)                                                                              |                                                                                           |
| ⚠️ | modified | 0x07 | SYS_CTL7 (trickle charging current, threshold, and charging timeout settings)                                 | moved to 0x06 adress; now hane only current settings, which takes whole byte              |
| ⚠️ | modified | 0x09 | SYS_CTL9 (standby enable and low voltage setting)                                                             | En_BATlow_Set flag replaced by Standby (Enter standby mode immediately when not charging) |
| ⚠️ | modified | 0x0A | SYS_CTL10 (battery low voltage setting)                                                                       | lowest voltage now is 2.5V                                                                |
| ⚠️ | modified | 0x0C | SYS_CTL12 (output maximum power selection register)                                                           | different poder selection map: 30, 45, 60, 65, 100, 140 W                                 |
| ✅  | added    | 0x0D | SELECT_PDO (select charging PDO gear)                                                                         | set preferable charging voltage                                                           |
| ✅  | added    | 0x2C | TypeC_CTL18 (PDO plus 10mA current enable, needs to be configured together with the current setting register) | ???                                                                                       |
| ❌  | deleted  | 0x30 | SOC_CAP_DATA (battery level data register)                                                                    |                                                                                           |
| ❌  | deleted  | 0x35 | MOS_STATE (input MOS status indication register)                                                              |                                                                                           |
| ✅  | added    | 0x35 | RECEIVED_PDO (receive PDO gear)                                                                               | actual charging voltage provided from charger                                             |
| ❌  | deleted  | 0x54 | IVbus_Sink_IADC_DAT0 (input current register)                                                                 |                                                                                           |
| ❌  | deleted  | 0x55 | IVbus_Sink_IADC_DAT1 (input current register)                                                                 |                                                                                           |
| ❌  | deleted  | 0x56 | IVbus_Src_IADC_DAT_Src_IADC_DAT0 (output current register)                                                    |                                                                                           |
| ❌  | deleted  | 0x57 | IVbus_Src_IADC_DAT1 (output current register)                                                                 |                                                                                           |
| ✅  | added    | 0x69 | TIMENODE1 (the first bit of the timestamp register) (the timestamp symbol is an ASCII character)              |                                                                                           |
| ✅  | added    | 0x6A | TIMENODE2 (the second bit of the timestamp register)                                                          |                                                                                           |
[... cut ...]
[IP2366 I2C regs V1.13 EN.pdf](https://github.com/user-attachments/files/16716922/IP2366.I2C.regs.V1.13.EN.pdf)

So, most functions are still compatible unless they use modified or deleted registers.
For example, 
```cpp
    void enableBatteryTypeSetting(bool enable);
    void setBatteryType(BatteryType type);
    void enableCurrentOrPowerSettingMode(bool enable);
    void setCurrentOrPowerSettingMode(CurrentSettingMode mode);
```
most likely they will not give any result, because they attempts to write 0x01(SYS_CTL1) register.

RWBRWB, 2024-10-26:
Do you know if you can get I2C access to the same LED pins on the IP2366 like you do on these same Ip2368 demo boards? 

I have a few of the same purple demo boards for the IP2366 with the LED indicators but thought I needed a special IP2366 chip to be able to communicate with it via I2C. Would be great to know if we could communicate via I2C with any of the IP2366 chips like your doing on the IP2368 chips.

D-314, 2024-10-30:
https://github.com/D-314/IP2366-Arduino-Library/tree/main

someone pls test and reply if it works

D-314, 2024-10-30:
@RWBRWB, try this solder points. Yiu have to remove resistors first.
[image]
[image]

RWBRWB, 2024-10-31:
It does not work. 

using proper 3v pull ups on the I2C lines you get no response on a I2C bus scan. 

If you don't use pullups you can get some false readings where it looks like it's responding but it won't consistently unless its a I2C chip version. The techs told me the same when I asked last year. 

Sure would be nice if the chip could do both but I guess it can't so they sell different versions.

D-314, 2024-10-31:
@RWBRWB, try to play with the INT pin (connect it to GND or VCC). Also try to turn off the chip completely (disconnect the charger, battery and 3v3 power supply), and then connect only the charger and the i2c bus (with INT connected to GND or VCC).

And what do you mean by "false readings"? Is there a device response at 0x75?

RWBRWB, 2024-10-31:
I'll add the I2C version of the chip and see if it works without any changes. 

The LED version will not provide I2C, that's been verified by multiple people selling the chips for Injonic. 

> @RWBRWB, try to play with the INT pin (connect it to GND or VCC). Also try to turn off the chip completely (disconnect the charger, battery and 3v3 power supply), and then connect only the charger and the i2c bus (with INT connected to GND or VCC). Also you have to slow down i2c bus to 100kHz for normal device operation.
> 
> And what do you mean by "false readings"? Is there a device response at 0x75?
> 
> > Have you tried this board yet?
> 
> I don't have this board, but I wrote a library because there were a lot of requests for this chip. So can you perform the tests under different conditions, please? I have published this library under the name "IP2366" in the Arduino library manager.
> Preferable tests
> 🟡 Yellow 	🟢 Green 	⚪ White 	⏛ Pull-Ups 	💡 Power
> ⚡VCC 	⏱ SCL 	📟 SDA 	Ω to VCC 	🔌 Cahrger
> ⏚ GND 	⏱ SCL 	📟 SDA 	Ω to VCC 	🔌 Cahrger
> ⚡VCC 	📟 SDA 	⏱ SCL 	Ω to VCC 	🔌 Cahrger
> ⏚ GND 	📟 SDA 	⏱ SCL 	Ω to VCC 	🔌 Cahrger
> ⚡VCC 	⏱ SCL 	📟 SDA 	❌ removed 	🔌 Cahrger
> ⏚ GND 	⏱ SCL 	📟 SDA 	❌ removed 	🔌 Cahrger
> ⚡VCC 	📟 SDA 	⏱ SCL 	❌ removed 	🔌 Cahrger
> ⏚ GND 	📟 SDA 	⏱ SCL 	❌ removed 	🔌 Cahrger
> ⚡VCC 	⏱ SCL 	📟 SDA 	Ω to VCC 	↪I2C, then charger
> ⏚ GND 	⏱ SCL 	📟 SDA 	Ω to VCC 	↪I2C, then charger
> ⚡VCC 	📟 SDA 	⏱ SCL 	Ω to VCC 	↪I2C, then charger
> ⏚ GND 	📟 SDA 	⏱ SCL 	Ω to VCC 	↪I2C, then charger
> ⚡VCC 	⏱ SCL 	📟 SDA 	❌ removed 	↪I2C, then charger
> ⏚ GND 	⏱ SCL 	📟 SDA 	❌ removed 	↪I2C, then charger
> ⚡VCC 	📟 SDA 	⏱ SCL 	❌ removed 	↪I2C, then charger
> ⏚ GND 	📟 SDA 	⏱ SCL 	❌ removed 	↪I2C, then charger

RWBRWB, 2024-11-02:
I swapped for the I2C version of the IP2366 chip and it works while the INT pin is floating after taking off the resistors. 

I tested your library and it works, the library needs cleaned up for this chip but that shouldn't be too much work as it looks like you have alot of it done already. 

Are you in the USA? I can send you a few of these I2C chips if you can get the library fixed up as I need it but have my hands full with multiple other things. 

[image]

[image]

[image]

So yea it works

RWBRWB, 2024-11-03:
I added the IP2366 Chip and it works as expected. 

Library gave me a good starting point also so thanks for that. 

[image]

[image]

D-314, 2024-11-04:
@RWBRWB, that's good!
Can you please provide detailed instructions with the actions that you have done? Or at least detailed photos with soldering points (I'm still not sure about the location of the SCL and SDA pins).
I also see you have pull-ups. What happens if you remove them?

RWBRWB, 2024-11-04:
Here is a picture showing which is SCL & SDA. 

[image]

3v Pull-ups are required for reliable I2C communication so even if you get it to work without pull-ups it's not recommended if you don't want random incorrect data at times. 

I have the INT pin attached to 3.3v so it never sleeps for testing and that works. 

After all that is wired up I just run your libraries ExtendedDataRead.ino and it works. I did adjust the getVsysPower function so the output is formatted properly and adjusted at tad for more accurate readings based on what is actually being consumed by the attached device. 

[image]

I haven't dug into the library to start exploring the other functions in it but look forward to. I wrote a library for the IP5839 chip in the past from scratch so nice to have a working library to start off with on the IP2366. If you do any more updates to this chip keep me updated as I will do the same if I add anything. 

I have 150 of these IP2366 - I2C chips I ordered from China and I could send you 5 for testing if have any boards to use them with. I see the IP2366 better than the Ip2368. 

> @RWBRWB, that's good! Can you please provide detailed instructions with the actions that you have done? Or at least detailed photos with soldering points (I'm still not sure about the location of the SCL and SDA pins). I also see you have pull-ups. What happens if you remove them?

D-314, 2024-11-19:
@RWBRWB, 🥺👉👈

RWBRWB, 2024-11-19:
> @RWBRWB, 🥺👉👈

Sorry bro, just been caught up running the biz which is leaving almost zero time for playing and testing unfortunately. 

I will report back soon as I get your library more, this weekend maybe. 

I did start laying out a custom PCB with 4 of the IP2366's on it. 2 for USB C and 2 for USB A ports for a battery pack I'm building.

MaBaTa, 2024-12-15:
> I did start laying out a custom PCB with 4 of the IP2366's on it. 2 for USB C and 2 for USB A ports for a battery pack I'm building.

I've also thought about this because I have a board that doesn't support I2C.
What kind of microcontroller are you having in mind for this?

Excited to hear about it...

RWBRWB, 2024-12-17:
if you hit the button to wake up the Ip2366 before plugging in your device does it work correctly then? 

It does the same to me when plugging in a Samsung Phone, if it's not woken up before plugging the phone in then it provides 5v vs 9v but if you wake up the chip or pull the INT pin to 3.3-5v to keep it from going back to sleep then it will supply the phone with the 9v quick charge voltage. 

Make sure you remove the I2C resistors also on the purple board. 

Mine work just fine after simply swapping chips. Make sure you got the correct I2C chips, hard to tell just by looking at them or the model numbers since they are just manufacturing data. 

> Hi, I have some of the 2366 purple boards. I bought the 2366 chip with I2C, but the board no longer works by default after the swap. When I plug in a PD charger, it only grabs 5V, and the normal voltage is 20V. Does the IP2366 need some initial programming over i2c to work as a power bank? I can only see the machine-translated datasheet, but I did not find it. I checked the swap was correct. Replaced again by the original 2366, the board works as expected, it is something with the new IC.

conradopoole, 2025-01-05:
> Sorry bro, just been caught up running the biz which is leaving almost zero time for playing and testing unfortunately.
>
>
>
> I will report back soon as I get your library more, this weekend [...]

Did you end up sending those PCBs for fabrication? I'd be curious to know if the reference schematic and the values are all you needed or if you had to make changes. 

I'm planning on building some custom PCBs too but still not sure if I'll use the IP2366 or the IP2368. I know you mentioned the 68 has a smaller BOM, for me it'd be charging a 4s1p 21700 battery so 140W is probably too much anyway.

MaBaTa, 2025-01-05:
> I'm planning on building some custom PCBs too but still not sure if I'll use the IP2366 or the IP2368. I know you mentioned the 68 has a smaller BOM, for me it'd be charging a 4s1p 21700 battery so 140W is probably too much anyway.

I would highly recomend you the IP2366, i have tested a lot with both and had a lot of compatibility and overheating Issues with the IP2368. I don't know if you guys have a thread for pro's and con's of the two, but i would like to share my experiences if wished...

Also you can configure the IP2366 to less then the maximum 140W to avoid overloading your batterys capacities.

conradopoole, 2025-01-05:
> I would highly recomend you the IP2366, i have tested a lot with both and had a lot of compatibility and overheating Issues with the IP2368. I don't know if you guys have a thread for pro's and con's of the two, but i would like to share my experiences if wished...

Thanks for the info. I guess I will just design around the IP2366 then, which is also a newer IC anyway. Did you just use the boards you can find on AliExpress or did you design a PCB yourself. I'm interested in finding out if the reference schematics in the datasheet work well.

RWBRWB, 2025-01-05:
I've been running the IP2366 for months now daily and it's ideal. 

If your going to power loads of 100+ watts then your going to need fan cooling. 

I run 140w loads on it with a 25x25x 5mm fan on it and the whole board runs cool to the touch with it blowing air down onto the inductor and fets. This will be the same for both chips so no getting away from it. 

The Ip2366 is a better design IMO, less parts to accomplish the same thing. 

The schematic is just fine to build off of, just keep it close to the demo board layout. 

OSHHardware the chinese site has alot of builds using these chips along with the scheamatic sheets you can look at for reference. 

I haven't found a better IC for USBC 140w sink source applications and I've looked everywhere over the last few years. 

> Thanks for the info. I guess I will just design around the IP2366 then, which is also a newer IC anyway. Did you just use the boards you can find on AliExpress or did you design a PCB yourself. I'm [...]

conradopoole, 2025-01-05:
> OSHHardware the chinese site has alot of builds using these chips along with the scheamatic sheets you can look at for reference.

Thanks. Whats the OSHHardware website? I know of the EasyEDA one, https://oshwlab.com/ but it doesn't seem to have any when you search for IP2366 or IP2368

RWBRWB, 2025-01-06:
https://oshwhub.com/wqlll/ip2366

> > OSHHardware the chinese site has alot of builds using these chips along with the scheamatic sheets you can look at for reference.
> 
> Thanks. Whats the OSHHardware website? I know of the EasyEDA one, https://oshwlab.com/ but it doesn't seem to have any when you search for IP2366 or IP2368

conradopoole, 2025-01-07:
> https://oshwhub.com/wqlll/ip2366

Thanks !! Did you end up getting rid of the Q5 that is in the VBUS line, that the datasheet says to only use for the certification, but then short? I just wonder if we can short VBUSI and VBUS and leave VBUSG floating/NC.

RWBRWB, 2025-01-07:
I left that FET out of my designs the same as the demo boards did.

Just leave the VBUSI & VBUS pins floating and yes just remove the FET and connect directly to the VBUS+ on the USBC connector. 

> Thanks !! Did you end up getting rid of the Q5 that is in the VBUS line, that the datasheet says to only use for the certification, but then short? I just wonder if we can short VBUSI and VBUS and [...]

conradopoole, 2025-01-07:
> Just leave the VBUSI & VBUS pins floating and yes just remove the FET and connect directly to the VBUS+ on the USBC connector.

I guess VBUS, VBUSI and even VBUSG can be left floating right? 

[image]

BTW I am putting the reference schematic in a KiCad file I can share once done.

conradopoole, 2025-01-07:
This is what I came up with, I think I didn't miss any mistakes copying over the datasheet. I have made a couple changes like removing the that Q5 FET, but for the most part it should be equal, I even tried to keep the reference numbers the same.

[image]

RWBRWB, 2025-01-07:
Here is what I created:

[image]

> This is what I came up with, I think I didn't miss any mistakes copying over the datasheet. I have made a couple changes like removing the that Q5 FET, but for the most part it should be equal, I even tried to keep the reference numbers the same.
> 
> If you guys think it's correct I can share the kicad source file
> 
> ![image](https://private-user-images.githubusercontent.com/5704796/400888755-03040f57-3fed-4b05-a8f5-c7cb6accce28.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNzc0ODIsIm5iZiI6MTczNjI3NzE4MiwicGF0aCI6Ii81NzA0Nzk2LzQwMDg4ODc1NS0wMzA0MGY1Ny0zZmVkLTRiMDUtYThmNS1jN2NiNmFjY2NlMjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDdUMTkxMzAyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjZiMDBiMGY0ZWFkNTZlYmUzYzg2ZmFjY2M4NDYyODMwNGEwNWE4ZDhmN2FhNGM2OTJkODM2MWFhMzhhYWE4NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3k5T_4EJjLxjdSOSPQEka2WiN4ogzpD4Sf_vFFb2Rb8)

conradopoole, 2025-01-07:
@RWBRWB Yours looks a bit nicer with more space to make it more readable, I'm gonna change it a bit like that. Also, I noticed you connected VBUS and VBUSI to VBUS, and only left VBUSG floating.

What FETs did you end up choosing that you know work well since you have already tested?

I will make some changes and share the KiCad file which is the LED version, do you wanna do the same with the I2C version? That way we have both versions for people to reuse in their projects. 

A couple questions about the I2C one. Do you know if it still works with LEDs or those GPIOs can only be used for I2C if using the I2C chip version? If there is no I2C communication, does it bootup with the resistor values for cell count, power and cell type/voltage ?

FilosofZ, 2026-10-13:
Небольшая информация. Актуальных i2c версий чипа ip2366 две: IP2366_I2C_D_STB и IP2366_I2C_D_ENP; первая в режиме ожидания потребляет 10мкА, но стартует только с кнопки либо по команде EN; вторая потребляет 100мкА, но стартует и автоматически по подключении кабеля + кнопкой или EN. 
[Ревизии ip2366.pdf](https://github.com/user-attachments/files/33384290/ip2366.pdf) 

Взял для экспериментов фиолетовую плату со светодиодами, после перепайки чипа на i2c стала перезагружаться каждые 5 сек, оказалось пин KEY висел в воздухе; я притянул его резистором 10кОм на gnd - все заработало.