D-314/IP2368-Arduino-Library #7: When I compile with the examples program "SimpleDataRead"  I get the following error

Jizyjiang, 2024-11-15:
FQBN: arduino:avr:uno
Using board 'uno' from platform in folder: /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6
Using core 'arduino' from platform in folder: /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6

Detecting libraries used...
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp -o /dev/null
Alternatives for IP2366.h: [IP2366@1.1.1]
ResolveLibrary(IP2366.h)
  -> candidates: [IP2366@1.1.1]
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp -o /dev/null
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src/Wire.cpp -o /dev/null
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src/utility/twi.c -o /dev/null
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /Users/user/Documents/Arduino/libraries/IP2366/src/IP2366.cpp -o /dev/null
Generating function prototypes...
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp -o /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/1672887249/sketch_merged.cpp
/Users/user/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/1672887249/sketch_merged.cpp
Compiling sketch...
/Users/user/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire/src -I/Users/user/Documents/Arduino/libraries/IP2366/src /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp -o /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/arduino/sketches/F152AC7BF3691C100FC0F217E1BB8CBD/sketch/SimpleDataRead.ino.cpp.o
In file included from /private/var/folders/ls/p18j8q3j12v3rdbjtt0mj7hm0000gn/T/.arduinoIDE-unsaved20241011-6166-fpttis.4do3a/SimpleDataRead/SimpleDataRead.ino:4:0:
/Users/user/Documents/Arduino/libraries/IP2366/src/IP2366.h:198:10: error: default argument missing for parameter 2 of 'void IP2366::setChargingPDOmode(IP2366::ChargingPDOmode, uint8_t*)'
     void setChargingPDOmode(ChargingPDOmode mode = ChargingPDOmode::V20, uint8_t * errorCode);
          ^~~~~~~~~~~~~~~~~~
/Users/user/Documents/Arduino/libraries/IP2366/src/IP2366.h:229:10: error: default argument missing for parameter 6 of 'void IP2366::enableSrcPdoAdd10mA(bool, bool, bool, bool, bool, uint8_t*)'
     void enableSrcPdoAdd10mA(bool en5VPdoAdd10mA = true, bool en9VPdoAdd10mA = true, bool en12VPdoAdd10mA = true,
          ^~~~~~~~~~~~~~~~~~~

Using library Wire at version 1.0 in folder: /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/libraries/Wire 
Using library IP2366 at version 1.1.1 in folder: /Users/user/Documents/Arduino/libraries/IP2366 
exit status 1

Compilation error: exit status 1

D-314, 2024-11-15:
@rmarket6, now shold be fixed in v1.2.0. Pls check again.

About Wire library: I tested the libraries on esp32. However prototypes for Wire methods should be tha same.

The problem was that i missed `errorCode = nullptr` default value for 2nd optional argument.

D-314, 2024-11-17:
@rmarket6, @Jizyjiang, pls use IP2368 library from arduino library manager with `IP2366.h` header. Separate IP2366 library is discontinued (because version from this repo aleready supports all functions from both IP2366 and IP2368)

closed 2024-11-17 by D-314