D-314/IP2368-Arduino-Library #10: Ip2366 on this board

sheedyaja, 2024-12-10:
[image]
[image]

This is a IP2366 based power/charger module
Any possibilities for this codes to work on that board?

D-314, 2024-12-12:
@pls share BOM file. It is difficult to see the chip label in the photo you provided

D-314, 2024-12-13:
@sheedyaja, okay, based on the components, the schematic seems similar to the IP2366. You can desolder these resistors (see the attached image) and use the points where they were connected as `I2C_SCL`, `I2C_SDA`, and `I2C_INT`

[image]

However, be cautious—this might be a revision of the IP2366 that does not support I2C (previously discussed in [issue #3](https://github.com/D-314/IP2368-Arduino-Library/issues/3)).  

That said, if you're lucky, you can use this library with your board.

sheedyaja, 2024-12-13:
@D-314 will try tomorrow, thanks

sheedyaja, 2024-12-15:
@D-314 update, it doesn't work

D-314, 2024-12-15:
@sheedyaja, could you please share a photo with your connections?

closed 2024-12-23 by D-314