D-314/IP2368-Arduino-Library #11: Library Request for STM32

MordFIdel, 2025-01-31:
Hi how can I adapt the library for st MCUs?

D-314, 2025-02-16:
Hi, @MordFIdel.

To adapt the library for ST MCUs, follow these steps:

### 1. Replace Wire Library
In the `IP236x.h` file, replace:
```cpp
#include <Wire.h>
```
with:
```cpp
#include <stm32****_hal_i2c.h> // Adjust based on your MCU
```

---

### 2. Modify Constructor
In the `IP236x.h` file, replace the constructor and member declaration:
```cpp
TwoWire *_wire;
IP236x(uint8_t address = 0x75, TwoWire &wire = Wire)
    : _address(address), _wire(&wire) {}
```
with:
```cpp
I2C_HandleTypeDef *_hi2c;
IP236x(uint8_t address = 0x75, I2C_HandleTypeDef *hi2c)
    : _address(address), _hi2c(hi2c) {}
```

---

### 3. Add Missing Bit Manipulation Macros
In `IP236x.h`, add:
```cpp
#define bitRead(value, bit) (((value) >> (bit)) & 0x01)
#define bitSet(value, bit) ((value) |= (1UL << (bit)))
#define bitClear(value, bit) ((value) &= ~(1UL << (bit)))
#define bitToggle(value, bit) ((value) ^= (1UL << (bit)))
#define bitWrite(value, bit, bitvalue) ((bitvalue) ? bitSet((value), (bit)) : bitClear((value), (bit)))
```
---

### 4. Rewrite I2C Methods
In `IP236x.сpp`, update the following methods according to your MCU's HAL I2C API:
- `void IP236x::begin()`
- `uint8_t IP236x::writeRegister(uint8_t regAddress, uint8_t value, uint8_t *errorCode)`
- `uint8_t IP236x::readRegister(uint8_t regAddress, uint8_t *errorCode)`

These functions should use `HAL_I2C_Master_Transmit` and `HAL_I2C_Master_Receive`.

---

### 5. Remove Arduino Dependency
Remove the following line from all `*.cpp` files:
```cpp
#include <Arduino.h>
```

After these changes, the library should be compatible with ST MCUs using HAL-based I2C communication.

vvvtj, 2025-02-28:
Вот библиотеки для stm 
https://gitee.com/Atonidas/ip2368-power-bank

closed 2025-06-24 by D-314